microelly2/Animation #7: Doku

herrdeh, 2015-05-27:
A really tiny little bit of doku would be nice. Or some hint to a tutorial. Just a few lines how to make it work. For the dummies.
Thanks!

microelly2, 2015-05-28:
see my homepage 
http://freecadbuch.de/doku.php?id=Animation%20Wokbench
and search for posts in the blog

herrdeh, 2015-06-01:
Its muchmuch better now. Now it starts to be useable for dummies as well!
I'd like to suggest to add these sentences into the readme:

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
First steps: 
* Create a new manager
* Put a mover or rotator under ("into") the manager
* Adjust controls in both
* Click the "Run Manager" button

For details see:
(links to videos here)
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

Belive me - a dummie like I am will save hours of useless research by such few sentences.