microelly2/Animation #10: Animation doesn't work on part imported with the assembly2 workbech

QiangF, 2015-10-31:
Hello,
I get this error when tring to animate a part impored with the assembly2 workbench.

    Running the Python command 'A_Runner' failed:
    Traceback (most recent call last):
      File "/home/q/.FreeCAD/Mod/Animation/Animation.py", line 2015, in Activated
        reinit()
      File "/home/q/.FreeCAD/Mod/Animation/Animation.py", line 1956, in reinit
        obj.Proxy.__init__(obj)
    
    Proxy_importPart instance has no attribute '__init__'

I also posted this on the forum.
http://forum.freecadweb.org/viewtopic.php?f=3&t=13002&p=104587#p104587

Thank you for your help!

OS: Debian GNU/Linux 8.0 (jessie)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.5841 (Git)
Build type: Release
Branch: master
Hash: e7d16c5f620bc4ddded8b8eae97d6e29c0012490
Python version: 2.7.9
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.7.0

microelly2, 2015-11-02:
I have already found the solution for this problem, but there is still another problem open - when assemby2  cannot solve the constraints - I will publish the next version when I have found a workaround/solution for that problem too.

QiangF, 2015-11-02:
From the user's perspect of view, maybe just set the auto update to false in the assembly2 module. Animation should always be done after the assembly is finished. The assembly2 is too slow for real time constraint solving, perhaps we should let the assembly2 workbech only set the initial position.

microelly2, 2015-11-02:
My idea is to run animation and assemby in parallel and store the placements into a file (bake the motion in Blender speach). Then the stored data can be used to run a realtime animation for demonstration purposes.

microelly2, 2015-11-02:
I have updated the Animation.py
It works for me.

closed 2015-11-02 by microelly2