microelly2/Animation #12: Problem using Dwire or DLine as motion path

DaveC49, 2016-01-01:
Hi microelly2,

Während der Erkundung der Animationswerkbank, habe ich schriftlich eine englische Dokumentation dessen, was ich in der Lage, von der Erforschung ab DasFreeCADBuch zu lernen. Wenn Sie möchten, kann ich für die Aufnahme zur Verfügung , dass mit der Werkbank zu machen.

Ich habe in der Lage, entweder eine Skizze oder einen Vektor nach den Bewegungspfad mit dem Mover -Objekt verwenden , haben aber kein Glück entweder mit einem DLine oder Dwire oder Form .edge für den Pfad . Ich denke, diese können noch nicht umgesetzt werden, insbesondere die shape.edge so scheint es keinen Weg, um die Kanten man folgen wollen auszuwählen. Zur Info , bekomme ich folgende Meldungen beim Versuch, mit einem Dwire und Dline als Pfad auszuführen.

Ich bekomme die folgende Traceback für eine Dwire :
Pfad bereits berechnet
************************* manager run loop:1/101  1
Running the Python command 'A_Runner' failed:
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/Animation/Animation.py", line 2054, in Activated
    FreeCADGui.doCommand("tt.Proxy.run(-1)")
  File "<string>", line 1, in <module>
  File "/usr/lib/freecad/Mod/Animation/Animation.py", line 1689, in run
    ob.Proxy.step(nw)
  File "/usr/lib/freecad/Mod/Animation/Animation.py", line 772, in step
    v=t.pop()

pop from empty list

und das folgende eine Dline :
Pfad bereits berechnet
************************* manager run loop:1/101  1
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/Animation/Animation.py", line 1815, in on_pushButton_clicked
    self.fun(self.vobj)
  File "/usr/lib/freecad/Mod/Animation/Animation.py", line 1880, in runManager
    tt.Proxy.run()
  File "/usr/lib/freecad/Mod/Animation/Animation.py", line 1689, in run
    ob.Proxy.step(nw)
  File "/usr/lib/freecad/Mod/Animation/Animation.py", line 772, in step
    v=t.pop()
IndexError: pop from empty list

Wenn mein Deutsch ist nicht sehr gut, Google zu übersetzen ist schuld !

Prost

David Cousens

microelly2, 2016-01-02:
Hallo David,
I'm working on a redesign of the workbench,
to get a animation along a path there is now a Pather, 
you can check out the  dev2 branch
and the testcase https://github.com/microelly2/Animation/blob/dev2/testcases/test_pather.py

I will test it for the other Draft curves next. 

To get nonlinear time behaviour it can be combined with the Speeder,
https://github.com/microelly2/Animation/blob/dev2/testcases/test_speeder.py
So it will be mopre general than the old Mover.
Documentation is in progress ....

microelly2, 2016-09-10:
It worked on my testcases, so maybe your dwire was empty?

closed 2016-09-10 by microelly2