microelly2/Animation #23: ER: Additional View for photographer

BPLRFE, 2016-09-11:
At the moment only fixed standard views could be chosen in the photographer.
Perhaps you could add a possibility to add also the actual camera position and orientation ?
At the moment the workaround would be to use Irfanview or something similar, make a screenshot of
every frame and put everything together into an animated gif using gif-x or something similar.

microelly2, 2016-09-11:
Yes, these 22,23 makes sence. I will rethink the image interface for camera configuration, preview and image post processing.

microelly2, 2016-09-12:
use the viewpoint node to configure camera position and direction
http://freecadbuch.de/doku.php?id=animation:viewpoint

closed 2016-09-12 by microelly2