microelly2/Animation #32: name 'reload' is not defined

sheinz, 2019-04-10:
I had an error on Arch Linux:
```
Running the Python command 'Anim_Placer' failed:
Traceback (most recent call last):
  File "<string>", line 74, in Activated
  File "<string>", line 1, in <module>

name 'reload' is not defined
```
I fixed it adding the following line in InitGui.py at line 74.
```
FreeCADGui.doCommand("from importlib import reload")
```
I haven't tested it with other versions of Python.

luzpaz, 2019-09-27:
Another usecase: source: https://forum.freecadweb.org/viewtopic.php?f=3&t=39668  
Most likely is a python3 issue.

unelsson, 2019-09-27:
`FreeCADGui.doCommand("from importlib import reload")` does fix this specific problem, but after fixing this I'm running into other problems: (`AttributeError: '_ViewProviderRotator' object has no attribute 'obj2'
Traceback (most recent call last):
  File "/home/nelsson/.FreeCAD/Mod/animation/Animation.py", line 256, in animforward`

It sounds to me like there are other python3 problems at play here?

luzpaz, 2019-09-27:
for sure, @microelly2 hasn't finished migrating this workbench yet

microelly2, 2019-10-14:
closed with 63f8d95a01a1d3d9f71606666d646f6e5a40a84d

closed 2019-10-14 by microelly2

luzpaz, 2019-10-14:
@microelly2 JFYI, you can use `Closes #32` in the git summary (not necessarily in the title) and it will auto-close this ticket and mention/link the commit hash