microelly2/Animation #35: After install from Freecad Addon manager, animation wb cannot be found in wb selector

bzb2, 2019-10-15:
I installed the latest animation workbench using freecad's addon manager on Oct 14,2019. After installation, workbench selector dropdown does not show an option for animation workbench.

The freecad used is...
OS: Ubuntu 18.04.3 LTS (LXDE/Lubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18353 (Git) AppImage
Build type: Release
Branch: master
Hash: 5de6763ed5889bfec76994e77d9c71b362448790
Python version: 3.7.3
Qt version: 5.12.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/Canada (en_CA)

snelweg, 2019-10-29:
Did you read the information on the frontpage or the info in the addonmanager?

[image]

shelby-cobra, 2019-10-30:
i had the same issue on Archlinux, error message:
`During initialization the error 'backend.qt4 is not a valid rc parameter (see rcParams.keys() for a list of valid parameters)' occurred in /home/andreas/.FreeCAD/Mod/animation/InitGui.py
Please look into the log file for further information`
Following [https://scipy-cookbook.readthedocs.io/items/Matplotlib_PySide.html](url) i commented out line 30 in Animation.py:
`# matplotlib.rcParams['backend.qt4']='PySide'`
and animation workbench is shown and can be used now.

bzb2, 2019-11-29:
Should your correction be implemented as a fix?

linuxuser63, 2020-04-20:
I am using :+1: OS: Mageia 7 (KDE//usr/share/xsessions/01plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.20514 (Git) AppImage
Build type: Release
Branch: master
Hash: a05ffd499366ab12fa1b43f37db73f4f1e9c332d
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
And th fix suggested by shelly-cobra did not work.
The animation workbench does not show up.

bpatrikm, 2020-09-02:
I found that replacing Qt4/qt4 with Qt5/qt5 everywhere, in addition to changing
`matplotlib.rcParams['backend.qt4']='PySide'`
to
`matplotlib.rcParams['backend']='Qt5Agg'`
was enough to get it started. I'm not sure if that backend is ever used, or if the line could just be commented out, but it's not triggering any errors.
(Obviously, I am using Qt version 5 rather than 4.)