microelly2/Animation #36: backend.qt4 is not a valid rc parameter (see rcParams.keys() for a list of valid parameters)'" occurred in /home/xx/.FreeCAD/Mod/Animation/InitGui.py

snelweg, 2019-10-30:
During initialization the error "'backend.qt4 is not a valid rc parameter (see rcParams.keys() for a list of valid parameters)'" occurred in /home/ton/.FreeCAD/Mod/Animation/InitGui.py
Please look into the log file for further information
Traceback (most recent call last):
  File "<string>", line 20, in <module>
  File "<string>", line 15, in __init__
  File "<string>", line 7, in __init__
  File "/tmp/.mount_FreeCAPXhj6V/usr/Mod/Start/StartPage/StartPage.py", line 302, in handle
    with open(qssfile, 'r') as f:
<class 'FileNotFoundError'>: [Errno 2] No such file or directory: '/tmp/.mount_FreeCAQQJwQP/usr/share/Gui/Stylesheets/Light-orange.qss'
[Errno 2] No such file or directory: '/tmp/.mount_FreeCAQQJwQP/usr/share/Gui/Stylesheets/Light-orange.qss'

`OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18614 (Git) AppImage
Build type: Release
Branch: master
Hash: efd55884ad7d40e77571a60c5bb906bbd8d1cbbd
Python version: 3.7.3
Qt version: 5.12.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)
`

shelby-cobra, 2019-10-30:
had the same isuue.
Solution see my comment here: [https://github.com/microelly2/Animation/issues/35](url)

snelweg, 2019-10-31:
ok, thank you. It worked.

(I copied the solution here for others to reference)

`i had the same issue on Archlinux, error message:
During initialization the error 'backend.qt4 is not a valid rc parameter (see rcParams.keys() for a list of valid parameters)' occurred in /home/andreas/.FreeCAD/Mod/animation/InitGui.py Please look into the log file for further information
Following https://scipy-cookbook.readthedocs.io/items/Matplotlib_PySide.html i commented out line 30 in Animation.py:
# matplotlib.rcParams['backend.qt4']='PySide'
and animation workbench is shown and can be used now.`

closed 2019-10-31 by snelweg